mathertel/DMXSerial #1: 32 bit code

PaulStoffregen, 2015-07-13:
Would you consider merging a large patch to support non-AVR chips?

I regularly hear requests to port your library to Teensy 3.1.  Already DMX transmit and receive code exists for Teensy 3.1, but not wrapped up in a nice library like yours.

So I'd like to ask, would you prefer to keep this very AVR specific (and non-AVR chips have their own libraries), or would you be open to merging fairly large contributions that duplicate quite a lot of the AVR specific stuff for non-AVR chips?

mathertel, 2015-07-14:
That is a really good idea to go 32 bit some time. Please let me know your implementation details so I can have a look how to support.

daniel3514, 2015-07-15:
Greetings, thank you for your interest
I use the library for control by DMX is developed by Deskontrol
http://www.deskontrol.net/blog/libreria-arduino-cuatro-universos-dmx/
only did the implementation of that library to this system without modification
this library has some problems when only one DMX output is required, but generally works very well

s-light, 2018-11-26:
is there any work in this kind of direction done or planed?

the M0 and M4 boards from Adafruit would also be a nice ARM target to play with dmx...

hbdsklf, 2023-03-02:
May be a bit out of scope but I just got it working on a Raspberry Pi 3B on Raspbian using a shield from Bitwizard. As soon as I could test it a bit more and find the time to clean up the code I'd be happy to make a pull-requests with the necessary patches (including the code for Raspberry or just the patches necessary to the existing code) if this is in the interest of @mathertel ..?