mathertel/DMXSerial #3: Is there a bug in the shield design?

alexkamp, 2016-01-17:
hey, 

I had a look at http://www.mathertel.de/Arduino/DMXShield/DMXShieldSchema.png

What makes me wonder is how OK1 and OK2 are connected. It looks to me like the arduino-side (left in the schema) is just turning a led on and off (even if the LED is inside the optocouplers). However, vanilla-led wiring (as in the simple example: https://www.arduino.cc/en/Tutorial/Blink) says: 

D2 -> Resistor -> LED -> GND
your design does
D2 -> Resistor -> LED -> 5V

Given that D2 provides 5V, connecting it to another 5V looks fishy. Is this a bug in your design?

maehem, 2016-01-17:
He is essentially driving the LED using Open Collector.  See this explanation and comparison here:
http://electronics.stackexchange.com/questions/44923/driving-a-led-with-an-open-drain-digital-input

You would write a zero to turn the LED on and 1 to turn it off.  So it is the inverse of the other method.

mathertel, 2016-01-18:
Thanks maehem for the comment and reference.
... and YES, it works as designed.

closed 2016-01-18 by mathertel

campino, 2016-01-18:
So I learned something. Thanks.