mathertel/DMXSerial #4: DMXSerial not working with no 512 DMX devices

noisetteofaddedia, 2016-06-03:
Hello
The library does not work with some lighting desks who doesn't send 512 channels
for example : the starway "commander 24" (24 dmx channels) or starway chaman (196 dmx channels)

imayoda, 2016-07-31:
you could try editing .h file parameter DMXSERIAL_MAX to narrow a bit the maximum channels scan.. 
if it work would be very usefull for low ram USART mcus.

mathertel, 2016-07-31:
In the standard implementations all received channels will be saved to memory. This is because the DMXSerial library doesn't know which of them you are interested in. This is by current design.

However you can change that and store only the channels you need by adding a parameter to the initialization.

closed 2017-10-29 by mathertel