mathertel/DMXSerial #5: RC oscillator 8Mhz support

imayoda, 2016-07-31:
Just asking, as per title, (with good manual freq. calibration) would it be possible to support no crystall operation in reduced frequency? 
Maybe there's still enough time to receive some dmx channels considering irq based apporach used in your lib.
BTW kudos

mathertel, 2016-07-31:
The time for the IRQ routine is indeed one problem you have to check but you also need time in your program as well.
250.000 baud with 8 MHz can be done. See http://wormfood.net/avrbaudcalc.php for details.
I propose that you gibe it a try.

imayoda, 2016-08-03:
Will try and post back if I get results, thanks a lot.

imayoda, 2016-08-09:
I'm back, excellent results with 8MHz internal osc.. 2 of 3 atmegas are working properly and stable with quite heavy receiving code and at least 4 analogWrite pwms active and looping fast. On the third chip a little OSCCAL fine setup fixed the problems (signal loss and blinkiness).
Very nice job you did with self defining serial setup.

mathertel, 2016-08-10:
Thanks @imayoda for this information and your testing. 
So it works on 8MHz ATMega boards without any modification to the software and you don't need the crystal and 2 caps. 
I hope the 8MHz is stable enough even when it gets warm or cold and adjusting the OSCCAL needs some equipment.
If you have any long time experience on this it would be great if you can comment that here.

closed 2016-08-10 by mathertel

imayoda, 2017-03-26:
Almost one year later followup, I report excellent performances @8MHz with bare rc osc..
Used absolutely original 328 and 328pu for 2 distinct dmx receiver projects, and they worked good in harsh env temperatures (10°C-40°C)..
Well done :)