mathertel/DMXSerial #8: 32bit DMXSerial for Teensy 3.2

henryflorence, 2016-12-23:
I am interested in using the dmx out, in and rdm capabilities of your libraries, as these currently seem to be the most complete. I am currently developing on arduino, but looking to port to teensy to utilise the increased resources.

Do you have any intention of porting to this platform? Or could you offer some guidelines as to how the task could be achieved. I noticed what looks like some of the DMXSerial code in DMXSimple which currently supports Teensy, but only for output.

s-light, 2018-11-26:
could be a duplicate of #1

mathertel, 2020-02-26:
With version 1.5 there is a processor specific layer that allows supporting multiple processor architectures. Please have a look if you like to try implementing Teensy support.

mathertel, 2021-03-08:
For Teensy there is another project that may fit: https://github.com/ssilverman/TeensyDMX

closed 2021-03-08 by mathertel