mathertel/DMXSerial #9: Working with Mini Pro or Nano

worldhugo, 2017-03-24:
Hi,

Thanks for all the work, it's great with Arduino Uno or Mega 2560.

I try to test your library with a Arduino Pro mini, but it's not working. Normal ?

And with a Arduino Nano ?

Thanks,

Hugo

jwhiddon, 2017-03-26:
I don't have that hardware, so I'm unable to test it.  Please report success or failure.

jwhiddon, 2017-03-26:
I don't have that hardware available to test.  What error are you getting?

mathertel, 2017-03-26:
What exact board (3.3V / 5V) are you using and what DMX level converter ?

worldhugo, 2017-03-27:
Thanks for replies !

I use a Mini Pro 3.3V / 8MHz ....

I think I have timing problem, because movements are jerky on my DMX equipment .... But I have not yet received my oscilloscope to measure that ..... 

It's possible the issue is the 8MHz board ?

imayoda, 2017-03-27:
the problem could be the level converting, if you're using plain rs485.. for 3,3V you need different transceivers

worldhugo, 2017-03-27:
Hi,

I think no, because I use DMX with teensy (3.3V logic out) and SN75176, and it's good ! :(

imayoda, 2017-03-27:
SN75176 is a 5V chip as per datasheet, it will give you weird behaviour at supply voltages lower than 4.75 (just like plain max485).. i did manage to use at 3.3V some max chips,  but in a very unpredictable behaviour :)

closed 2017-10-29 by mathertel