mathertel/DMXSerial #11: Reducing memory usage of _dmxData

njh, 2017-08-28:
Hello,

Thank you very much for writing this library. I have been using DMXSerial very successfully for a couple of years with my home lighting controller:
https://github.com/njh/ihlc

We turn our home lighting on and off daily using your library!

I have been looking at extending it to be controlled via IPv6, using a library I wrote called [EtherSia](http://github.com/njh/EtherSia). Together the two libraries use up quite a lot of the Uno's RAM - and I get warnings in the IDE.

I am currently only using around 20 channels.

Have you thought of any way of changing the memory usage for the DMX data?

```C++
#define DMXSERIAL_MAX 512 // max. number of supported DMX data channels

uint8_t  _dmxData[DMXSERIAL_MAX+1];
```

Adding `malloc()` increases the programme memory size - and not sure dynamic memory management desirable. But I am not sure of a neat way of the sketch indicating how much static memory they would like to be allocated?

Thanks!

nick.

mathertel, 2017-09-01:
Hello,

this is the maximum array of course.
You can cut it reduce it to starting-address + channels-needed in the library.

I never used dynamic memory allocation.

Best Regards, Matthias

https://github.com/njh/ihlc

njh, 2017-09-01:
I was wondering if you could think of a way of achieving this, without custom modifications to the library - i.e. by passing in a parameter from the Sketch somehow.

As it is quite a generic question, maybe I will ask on the Arduino Stackoverflow.

TraylorHead, 2017-09-01:
You may want to check on using a 1284P chip. It has more memory for code and variables.

 

 

James Traylor

TraylorHead Logo-24mm-03

 

TraylorHead

Douglasville, Ga

(770) 313-5574

 

This email has been checked for viruses by AVG.
http://www.avg.com

closed 2017-10-28 by njh