mathertel/DMXSerial #16: Send & Receive mode

anklimov, 2017-11-11:
Hello
I'm using DMXSerial in my project couple of years and thank you for your effort!
My controller receiving DMX signal and sending simultaneously 
Unfortunately, that is not possible to use DMXSerial bi-directionally 
That was a reason, why I'm using DMXSerial for receiving and DMXSimple for sending, that ti not efficient so much.
Is it possible to add possibility to operate DMX in 2-way mode (with different buffers, of course)? If will be possible to share one UART for this - it might be brilliant

DetsCastelli, 2019-12-15:
I think the problem is, that the baud rate is changed to send a break signal. All bytes coming in during this time can then not be recognized correctly. Maybe it's possible to use a timer to send the break

mathertel, 2020-02-26:
not possible. You need multiple independent Serial interfaces to support this scenario.

closed 2020-02-26 by mathertel

Tejes, 2020-02-26:
The MEGA2560 has 3 hardware UARTs. I'm in the design phase of an application that would be a "man-in-the-middle" for a DMX bus, mostly forwarding everything that comes in, but altering a few addresses if set to. Is it really only possible using two different libraries for the same thing? I have Arduinos with 328p and MEGA2560, also a Raspberry Pi 3B+, but no luck with this topic on any of them yet. Could you recommend something? I'm more of a software guy than a hardware one.

glaing, 2020-02-27:
I'm also trying to achieve this scenario with DMX Serial and this shield https://www.amazon.com/Arduino-TinkerKit-Master-Shield-T140060/dp/B00EIB4ZF8

It's not working yet, but I'd be interested to hear how you've got on with your setup!