mathertel/DMXSerial #17: SoftwareSerial and DMXSerial

cstranex, 2017-12-01:
Is it possible to use other serial devices using SoftwareSerial while DMXSerial is running in receive mode?

Yona-Appletree, 2018-05-11:
I'm running into issues with that as well. Probably something to do with the interrupts? I'd love to know if this is possible.

arneboe, 2019-06-04:
Am running into problems as well, receiving only works on the first few channels when software serial is enabled.

mathertel, 2019-06-04:
As I know SoftwareSerial is disabling interrupts and will interfere with the DMX that relies on receiving the interrupts from the hardware based serial port.

When requiring another serial port please use an Arduino Mega or Leonardo that supports 2 ports in Hardware.

/Matthias Hertel/

arneboe, 2019-06-05:
thx for the info :-)

closed 2019-06-16 by mathertel