mathertel/DMXSerial #20: should work with Arduino UNO/Mega and Tinkerkit DMX Master Shield?

moebiussurfing, 2018-04-05:
... I am looking for an alternative to DmxSimple

oliverh57, 2018-04-05:
It works with the shield. You need to change the settings on the transceiver to make it work however. There are 3 solder pads that need to be connected to ground.

moebiussurfing, 2018-04-06:
thanks. and whats the differents with this library?
maybe that the arduino is detected by any dmx desktop software like a 'commercial interface'?

mathertel, 2018-04-06:
The DMXSerial library takes advantage of the Hardware Serial Interface of the Arduino chip. This reduces massively the CPU cycles required for the DMX protocol especially for receiving DMX.
The Arduino is giving standard 5V TTL level signals. The shield is required to transform this into the RS485 level signals as required by the DMX specification.
I implemented the library to make Arduino based implementations as robust as I needed it for my projects.

closed 2018-04-06 by mathertel

mathertel, 2018-04-21:
/Matthias Hertel/