mathertel/DMXSerial #26: not working with a nano

EgoVanautgaerden, 2019-02-07:
I tried the library on an arduino nano 5v with the dmx receiver example code.
When there is DMX data the RX led blinks up but the arduino doesn't process the packets.
The pwm outputs stay at the predefined level of 80,0,0 .
When I use a scope on the RX pin of the arduino I see the DMX data stream coming in and changing when I move a fader on the controller.

Is there a compatibility issue with the arduino nano?

Kinds regards,
Ego

imayoda, 2019-02-10:
silly question but needs to be made..
did you used a 485 transceiver in front of the nano's serial rx pin ?

EgoVanautgaerden, 2019-02-11:
No problem,

Yes I used a 485 transceiver connected according to this schematic:
http://www.mathertel.de/Arduino/DMXShield/DMXShieldSchema.png

When I said I used a scope, I meant directly measured on the RX pin off the arduino, after the 485 & optocoulers.

imayoda, 2019-02-11:
ok :)
check on the scope the efficiency of the nano's clock (xtal or timer referenced pwm from one pin @high freq)
I've had a problem on a pro from cinese manufacturer, xtal was wonky, also be shure to set the bootloader to at least 8MHz (better 16).
Dmx requires somewhat high precision

EgoVanautgaerden, 2019-02-18:
Short story: It works and there was nothing wrong with the library nor the arduino.

Long story: I checked the effiency of the clock and it was running at a stable 16 ish MHZ (not exactly 16 but close enough). 
Then I cross referenced with an official arduino and still nope.
While I was looking at the DMX signal on my scope something seemed odd, the break was gigantic long and the MAB was also quite on the long side, also the signal didn't seem to be stable...
Having thought that I was testing from a quite cheap DMX controller, I was doubting that it was actually the arduino.
Today I checked it with the Hog4 we have at the theatre and everything worked from the first try.
Needlessly to say this cheap DMX controller is going into the garbage.

Thank you very much for your help!

Kind regards,
Ego

imayoda, 2019-02-19:
happy you figured out..
but don't trash the controller, often it only has a nasty cheap xtal .. 2 minutes fix to get it on the right timing :D

closed 2019-06-16 by mathertel