mathertel/DMXSerial #27: DMXReceive works only after flashing on Arduino Pro Micro

dronus, 2019-02-19:
Using an Arduino Pro Micro (an unofficial clone based on ATMega32U4), the DMXReceive mode is only working once after flashing (and eratically after USB serial initalisation).

-if the Pro Micro starts up after flashing, the DMXReceive mode is working perfectly.
-if the Pro Micro starts by power up (with and without USB data connection), it does not work.
-if the Pro Micro is started by power up, but the USB serial device is initialized later (eg. by opening the Arduino IDE serial console for the first time after plugging), the DMX receiver sometimes(but rarely) start working. 
-This bug does not depend on wether Serial (over USB) is initialized or not

For me it looks like some UART initalization is missing for this platform, which seems to be done by the USB programming protocol. 

Any ideas?

dronus, 2019-02-19:
It also starts to receive about 5 seconds after manually triggering RST pin, if powered without USB data connection. But won't do so after powerup without triggering RST pin.

cfreeman, 2019-03-06:
I have had even less joy trying to get this work on an unofficial clone Pro Micro (ATMega32U4). 

My wiring is:
MAX485 P1 -> Arduino RX1
MAX485 P2 -> Arduino GND
MAX485 P3 -> Arduino GND
MAX485 P4
MAX485 P5 -> Arduino GND + DMX P3
MAX485 P6 -> DMX P2
MAX485 P7 -> DMX P3
MAX485 P8 -> Arduino Raw

and my sketch:
```C
#include <EEPROM.h>
#include <DMXSerial.h>
#include <Arduino.h>

void setup() {
  delay(5000);
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(9600);

  DMXSerial.maxChannel(3);
  DMXSerial.init(DMXReceiver);
}

void loop() {
  delay(1000);
  int8_t val = DMXSerial.read(1);
  Serial.print("CH1: ");
  Serial.println(val);
}
```

I send through a DMX command that sets the first channel to 255 (from a Enttec OpenDMX USB device). But the serial monitor keeps pumping out (CH1: 0).

I put a few Serial.println commands into ```ISR(USARTn_RX_vect)``` to try and find out what was going on. 

```C
ISR(USARTn_RX_vect)
{

  uint8_t  USARTstate = UCSRnA;    // get state before data!
  uint8_t  DmxByte    = UDRn;	   // get data
  uint8_t  DmxState   = _dmxRecvState;	//just load once from SRAM to increase speed

  Serial.print("ISR U(");
  Serial.print(USARTstate);
  Serial.print(")S(");
  Serial.print(DmxState);
  Serial.print(")D(");
  Serial.print(DmxByte);
  Serial.print(") - ");

  ...

  Serial.println("UNHANDLED");
} // ISR(USARTn_RX_vect)
```

When a message comes in, my serial monitor dumps this:
```
CH1: 0
ISR U(160)S(1)D(248)
ISR U(160)S(2)D(248) - UNHANDLED
ISR U(160)S(2)D(248) - UNHANDLED
ISR U(160)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(168)S(2)D(248) - UNHANDLED
ISR U(184)S(2)D(0)
CH1: 0
```

I'm guessing I have done something silly?

dronus, 2019-03-10:
I've tried several ATMega32U4 boards from different sources, and it turns out there are flaky ones I guess.
The ATMega32U4 platform has an integrated USB stack, that is configured by the bootloader. So the bootloader strongly affects actual operation even if it hands control over to the user program. 

I had several devices which on some occasion (eg. installing the interrupts or configuring timers like this library does) something goes wrong and the device enters a very slow state (maybe interrupt thrashing occurs). Any operation is slowed down as the device frequency where much lower, even `delay` periods gets longer. USB serial output still works, but slowed down too. All timing related things like DMX transfers of course do not. 
The buggy state is entered if running sketches using this lib after cold start. But normal operation can be restored by a warm start (hardware reset). So I would blame this on a buggy bootloader, nothing DMXSerial has to care about.

dronus, 2019-03-10:
Question is, how to sort out bad Arduino Pro Micro boards from cool ones without analyzing them by installing DMXSerial sketches...

closed 2019-03-10 by dronus