mathertel/DMXSerial #29: 2 Universes on a board

JLightMedia, 2019-04-29:
Does this library support to generate 2 DMX  universes at the same time?

imayoda, 2019-04-30:
yes, just use a dev board that exposes the tx1 pin..
you will also loose the possibility to have serial debug on

JLightMedia, 2019-04-30:
Hello! Thanks for answer. But how defined two different pins for out put and how to  describe two different universes  in a program code?

imayoda, 2019-04-30:
sorry pal, been confusing as i was reading the esp8266 thread but responded here thinking it was an esp8266 question.. my mistake, really sorry

mathertel, 2019-05-12:
No. Just one DMX universe.
There is only one Serial interface implemented and there is only one set of service routines for this interface. The interface can be selected by using pre-precessor directives.

closed 2019-05-12 by mathertel