mathertel/DMXSerial #33: multiple definition of `__vector_19'

JLightMedia, 2019-11-08:
Very often, when I try to compile the library, I got this kind of error. 

### **HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_18'
libraries/DMXSerial-master/DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_19'
libraries/DMXSerial-master/DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status**

mathertel, 2019-11-09:
You can use the Serial interface either for DMX or for Serial.print(…) but not both in the same script.
The Serial interface of the UNO chip is required to have a robust DMX signal and still enough power left for other processings.

There are some options you can use. They are explained in one of the posts linked at the end of the documentation: 

<http://mathertel.blogspot.com/2013/05/update-for-dmxserial-library.html>

closed 2019-11-09 by mathertel

JLightMedia, 2019-11-09:
Thank you for answer. But I don’t use serial interference in my code. I’ve used, but before use DMX Serial library I commented Serial. initializations.

mathertel, 2019-11-10:
Removing the initialization only is not enough, you have to remove all references.

But it may be even hidden: Have a look at https://github.com/mathertel/DMXSerial/issues/14 : There are libraries (here the SD library) that have a hard reference to the Serial built-in.

JLightMedia, 2019-11-10:
Hm, It's very interesting. Is there any other library that have strong references to the Serial interface ? 
In my my project I use libraries that work with network:
#include <Artnet.h>
#include <Ethernet.h>
#include <Udp.h>
#include <EEPROM.h>