mathertel/DMXSerial #35: ESP32 resp. Odroid-GO for DMX-receiver

gustyfeinen, 2020-02-04:
Hey
I try to build a small monitoring for a DMX-line, before creating a controller. Therefore a ODROID-GO would be a nice device with an DMX-USB cable at its input. I try to use the library
#include <DMXSerial.h>
but the compiler does not accept ... :
exit status 1
Error compiling for board ODROID ESP32.
Is there a work around for it ?

mathertel, 2020-02-04:
Hi,

when you look into DMXSerial.cpp you see that it uses direct register access and interrupt service routines to the ATMEL 186 and similar chips to get control on communications.
Today only these processors are supported but not the ESP32-WROOVER.

/ Matthias

Coddy2, 2023-08-22:
Hi, is there any news on ESP32 support? Thank you

thunermay, 2025-04-13:
I would also be glad :)
VG Leon