mathertel/DMXSerial #37: Support for atmega4809?

andrewga, 2020-02-14:
would there be future support for the atmega4809?

mathertel, 2020-02-14:
I am working on it. You can checkout the branch: https://github.com/mathertel/DMXSerial/tree/megaavr

If you find time to test it and five feedback would be phantastic.

andrewga, 2020-02-14:
I just tried it on the Arduino uno wifi and it works!  I'm trying to understand the code, I am using the DmxSerialSend.  I would like it to display just one color like 255,0,0.  How would I do that?  Again Awesome job!  I had been racking my head on getting it to work on the 4809!

mathertel, 2020-02-26:
Please update to the version 1.5.0 ff with support for the **megaavr4809** processor like the one in **Arduino Every** and **Arduino WiFi Rev. 2**.

closed 2020-02-26 by mathertel

andrewga, 2020-02-28:
Mathertel,  Its still not working correctly for me.  I changed the code to show random colors as follows it shows some colors and hangs on some and flickers.

void setup() {
   Serial.begin(9600);
  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH);
  DMXSerial.init(DMXController);
  
  DMXSerial.maxChannel (8);

  pinMode(RedPin,   OUTPUT); // sets the digital pin as output
  pinMode(GreenPin, OUTPUT);
  pinMode(BluePin,  OUTPUT);

  
} // setup

// loop through the rainbow colors 
void loop() {
  int x = random(255);
    int x1 = random(255);
    int x2 = random(255);
       DMXSerial.write(1, x);
       DMXSerial.write(2, x1);
       DMXSerial.write(3, x2);
       Serial.println(DMXSerial.read(1));
       Serial.println(DMXSerial.read(2));
       Serial.println(DMXSerial.read(3));
       
    analogWrite(RedPin,   x); 
    analogWrite(GreenPin, x1); 
    analogWrite(BluePin,  x2); 
      delay(1000);
} // loop