mathertel/DMXSerial #38: Support for shield CTC-DRA-10-R2

davidmu23, 2020-03-22:
I tried this on the CTC-DRA-10-R2 DMX Shield but it doesnt work correctly.  Any idea on how to get this working on this shield?  I'm also using the Arduino uno wifi.

mathertel, 2020-03-31:
I don't have this shield s I cannot check it. Maybe soneone can link the schema or has solved this.
I leave this issue open for a while...

davidmu23, 2020-04-02:
These are the only two links I can find regarding this shield:http://image.dfrobot.com/image/data/DFR0260/DMX%20Shield%20User%20manual.pdf 
http://dmxshield.blogspot.com/
I don't know if this can help you

andrewga, 2020-05-26:
heres some info I found, don't know if this can help
DMX SHIELD CTC-DRA-10-R2  
jumpers config:
EN(run) <==  EN¯ (updt fw) ! Please For update firmware disconect battery 7-12v only usb 3.5v or arduino can dead !
Slave   ==>  DE  (master)
TX-io   <==  TX-uart
RX-io   <==  RX-uart
Pins 2,3,4 are used to transmit DMX signals
xlr output to xlr input rgb

mathertel, 2020-05-27:
The DMXSerial library uses the lines TX, RX and pin 2 for direction. You have to set the jumpers right.
pin 2 is used to control the direction using HIGH for sending and LOW for receiving.
I haven't found a schema and I don't have a shield but I assume from some research that the Conceptinetics sources use the same levels. looks like being compatible.

andrewga, 2020-05-27:
I have the jumpers set to
EN
DE
TX-io
RX-io

here is how I have my setup:
`void setup() {
  Serial.begin(9600);
  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH);
  DMXSerial.init(DMXController,2);
  DMXSerial.maxChannel (8);
} `

jabadabado, 2020-05-27:
When using the ATMega2560 and the Serial2 library, and changing the UART to Port1, the PIN2, the direction of the 485 chip, the PIN2 is the TXD1 pin on the ATMega2560. So it doesn't work only by changing the define to ATMega2560 in Serial2. Changing pin 2 to another pin, does not seem to work. Maybe it needs to be changed elsewhere too? Any suggestions for testing on a ATMega2560?