mathertel/DMXSerial #41: Version 1.5.0 in GitHub differs from version from Arduino IDE library manager 1.5.0.

yunjova, 2020-04-30:
Hello,

The version 1.5.0 in GitHub differs from the version from Arduino IDE library manager 1.5.0.
This is probably because there are some updates in GitHub without an updated version in the library.properties file.

I assume that GitHub is correct (latest) (the difference is in the initialisation of the stop bits in the files DMXSerial_avr.h and DMXSerial_megaavr.h).

Question: Why are bytes with one stop bit now accepted (DXM standard is two stop bits)?

Great library by the way!

Kind regards,
Johan.

mathertel, 2020-04-30:
sometimes I do changes without doing a final test so Github is more advanced than the version from the library manager - not to confuse those people just using the release version.
When you like to support - as you do - you should use the github version (git clone into the library folder).

The DMX sender will always have to send 2 stop bytes. The receiver here already accepts the data after the first stop byte - not waiting for the second one - and just taking a small pause to the next byte. That's compatible on the receiving party.

closed 2020-04-30 by mathertel

yunjova, 2020-05-01:
Hi,

Just wondering: The IDE version uses **SERIAL_8N1** (and not **SERIAL_8N2**) in the below cases. Is this then correct/compatible with the DMX specification?

/// Initialize the Hardware UART serial port registers to the required mode.
void _DMX_setMode(DMXUARTMode mode)
{
...
  } else if (mode == DMXUARTMode::TDATA) {
...
    UCSRnC = **SERIAL_8N2**; // send with 2 stop bits for compatibility
  } else if (mode == DMXUARTMode::TDONE) {
...
    UCSRnC = **SERIAL_8N2**; // send with 2 stop bits for compatibility
  } // if
} // _DMX_setMode()

Kind regards.