mathertel/DMXSerial #43: megaavr PlatformsIO compability

jonolt, 2020-06-01:
compiling this simple working example for the Arduino Nano Every
```c++
#include <Arduino.h>

#include <DMXSerial.h>

void setup() {
  // put your setup code here, to run once:
  // Serial.begin(9600);
  DMXSerial.init(DMXController, 4);
}

void loop() {
  // put your main code here, to run repeatedly:
  // if(Serial.available()){}
}
```

works flawless with the Android IDE but fails at PlatformIO with

```
Processing nano_every (platform: atmelmegaavr; board: nano_every; framework: arduino)
---------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/nano_every.html
PLATFORM: Atmel megaAVR 1.0.0 > Arduino Nano Every
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES: 
 - framework-arduino-megaavr 1.8.6 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DMXSerial> 1.5.0
Building in release mode
...
Compiling .pio/build/nano_every/src/main.cpp.o
Compiling .pio/build/nano_every/lib001/DMXSerial_ID912/DMXSerial.cpp.o
Archiving .pio/build/nano_every/lib001/libDMXSerial_ID912.a
Indexing .pio/build/nano_every/lib001/libDMXSerial_ID912.a
Linking .pio/build/nano_every/firmware.elf
DMXSerial.cpp.o (symbol from plugin): In function `_DMX_init()':
(.text+0x0): multiple definition of `__vector_26'
.pio/build/nano_every/FrameworkArduino/UART1.cpp.o (symbol from plugin):(.text+0x0): first defined here
DMXSerial.cpp.o (symbol from plugin): In function `_DMX_init()':
(.text+0x0): multiple definition of `__vector_27'
.pio/build/nano_every/FrameworkArduino/UART1.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/nano_every/firmware.elf] Error 1
```

First the framework version was 1.8.5 (not 1.8.6) which caused some namespace errors which was fixed with an [update of the platformio library](https://github.com/platformio/platform-atmelmegaavr/issues/3), that I have requested. Now they should be identical and I'm getting above error.

I don't know C/C++ and avr-gcc well enough to solve this one alone, though I know the problem lies somewhere in the UART access.

I will/can also try to solve this myself if you point me in the right direction.

mathertel, 2020-06-04:
Hi Johannes, These 2 vectors are the interrupt routines for the Serial UART1. I also see no special thing pointing to something else using UART1. I don't use the -DAVR_NANO_4809_328MODE option but it compiles as well on Arduino without reporting conflicts.
This is odd. I wish you find "something".

jonolt, 2020-07-20:
### Identified Problem

After not getting any different results by playing with the compiler flags, i checked the files being linked. PlatformIO does -at least with my (default) settings- compile all files from the arduino core and tries to link them all with my other source files, even when I don't use them

```
Building in release mode
avr-g++ -o .pio/build/nano_every/firmware.elf -Os -flto -mmcu=atmega4809 -Wl,--gc-sections -Wl,--section-start=.text=0x0 -fuse-linker-plugin .pio/build/nano_every/FrameworkArduinoVariant/variant.c.o .pio/build/nano_every/FrameworkArduino/CDC.cpp.o .pio/build/nano_every/FrameworkArduino/NANO_compat.cpp.o .pio/build/nano_every/FrameworkArduino/Tone.cpp.o .pio/build/nano_every/FrameworkArduino/UART.cpp.o .pio/build/nano_every/FrameworkArduino/UART0.cpp.o .pio/build/nano_every/FrameworkArduino/UART1.cpp.o .pio/build/nano_every/FrameworkArduino/UART2.cpp.o .pio/build/nano_every/FrameworkArduino/UART3.cpp.o .pio/build/nano_every/FrameworkArduino/UNO_compat.cpp.o .pio/build/nano_every/FrameworkArduino/USBCore.cpp.o .pio/build/nano_every/FrameworkArduino/WInterrupts.cpp.o .pio/build/nano_every/FrameworkArduino/WMath.cpp.o .pio/build/nano_every/FrameworkArduino/abi.cpp.o .pio/build/nano_every/FrameworkArduino/api/Common.cpp.o .pio/build/nano_every/FrameworkArduino/api/IPAddress.cpp.o .pio/build/nano_every/FrameworkArduino/api/PluggableUSB.cpp.o .pio/build/nano_every/FrameworkArduino/api/Print.cpp.o .pio/build/nano_every/FrameworkArduino/api/Stream.cpp.o .pio/build/nano_every/FrameworkArduino/api/String.cpp.o .pio/build/nano_every/FrameworkArduino/hooks.c.o .pio/build/nano_every/FrameworkArduino/main.cpp.o .pio/build/nano_every/FrameworkArduino/new.cpp.o .pio/build/nano_every/FrameworkArduino/wiring.c.o .pio/build/nano_every/FrameworkArduino/wiring_analog.c.o .pio/build/nano_every/FrameworkArduino/wiring_digital.c.o .pio/build/nano_every/FrameworkArduino/wiring_pulse.S.o .pio/build/nano_every/FrameworkArduino/wiring_pulse.c.o .pio/build/nano_every/FrameworkArduino/wiring_shift.c.o .pio/build/nano_every/src/main.cpp.o -L.pio/build/nano_every -Wl,--start-group .pio/build/nano_every/lib4b0/libDMXSerial.a -lm -Wl,--end-group
DMXSerial.cpp.o (symbol from plugin): In function `_DMX_init()':
(.text+0x0): multiple definition of `__vector_26'
.pio/build/nano_every/FrameworkArduino/UART1.cpp.o (symbol from plugin):(.text+0x0): first defined here
DMXSerial.cpp.o (symbol from plugin): In function `_DMX_init()':
(.text+0x0): multiple definition of `__vector_27'
.pio/build/nano_every/FrameworkArduino/UART1.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/nano_every/firmware.elf] Error 1
```
while the ArduinoIDE is only linking the neccesarry files
```
Linking everything together...
/home/johannes/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mmcu=atmega4809 -o /tmp/arduino_build_176197/sketch_jul20a.ino.elf /tmp/arduino_build_176197/sketch/sketch_jul20a.ino.cpp.o /tmp/arduino_build_176197/libraries/DMXSerial/DMXSerial.cpp.o /tmp/arduino_build_176197/core/variant.c.o /tmp/arduino_build_176197/../arduino_cache_832754/core/core_arduino_megaavr_nona4809_mode_off_c1f597cf3dd931fc8d6bf7075b294c68.a -L/tmp/arduino_build_176197 -lm
/home/johannes/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O binary -R .eeprom /tmp/arduino_build_176197/sketch_jul20a.ino.elf /tmp/arduino_build_176197/sketch_jul20a.ino.bin
/home/johannes/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_176197/sketch_jul20a.ino.elf /tmp/arduino_build_176197/sketch_jul20a.ino.eep
/home/johannes/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_176197/sketch_jul20a.ino.elf /tmp/arduino_build_176197/sketch_jul20a.ino.hex
Using library DMXSerial at version 1.5.2 in folder: /home/johannes/Arduino/libraries/DMXSerial 
/home/johannes/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino_build_176197/sketch_jul20a.ino.elf
Sketch uses 2678 bytes (5%) of program storage space. Maximum is 49152 bytes.
Global variables use 560 bytes (9%) of dynamic memory, leaving 5584 bytes for local variables. Maximum is 6144 bytes.
```
(adding the UART1.cpp.o to the linking command of the ArduinoIDE will raise the same  error).

So it is no real bug since you usually shouldn't use Serial1 when using DMX Serial any ways. Now I only have to find out, how to solve this in PlatformIO. I will write an update when I have found a solution, until then I symlinked my source files into a ArduinoIDE folder strucure (and enable use external editor in the options) for my cross platform tests.

mathertel, 2021-06-08:
Thanks for the investigation.

closed 2021-06-08 by mathertel

ScottLeMagicien, 2024-08-15:
Hi Johannes!

I read with interest your investigation into the errors you were getting trying to compile a simple program using the DMXserial plugin.

I was wondering if you ever figured out a way to make it work with the platformio compiler?

Thanks!!
Kind regards,
Scott

jonolt, 2024-08-15:
Hi Scott,

I don't think I've come up with a solution, and I have not worked on the project since.
I just checked and my latest commits still have the workaround to program with paltformio and compile with the arduino-ide by symlinking the required files from the paltformio folder tree to a flat arduino folder as noted in my last post on this thread. 

I hope you will have more success in solving the problem properly.