mathertel/DMXSerial #45: Minimum supported inter-slot/-frame time when receiving?

septatrix, 2020-07-10:
Are there any experiences on what the minimum required time between two frames is to be correctly received? I know this is based on the processor speed so maybe you have some empirical values with certain processors / clock speeds.

For sending it is not to important as other devices should support up to a second of spacing though when I want to monitor other controllers it is rather important to correctly receive every frame...

mathertel, 2020-08-13:
as you said: no minimal gap defined. For High Speed applications this is also not wanted !

closed 2020-08-13 by mathertel

septatrix, 2020-08-17:
Yes I know there is no minimal inter-frame spacing though I think you misunderstood my question. If I run this library on a microcontroller and let it receive/monitor the signal how fast could the signal be? For example a software serial implementation can only work up to a specific speed until it misses frames and by the looks of it this library it uses some setup after each frame to jump back to a software implementation to recognize the start of a frame and then switches to UART. Do you have any experience regarding speed?

mathertel, 2021-06-08:
As receiving is completely interrupt driven there is no speed limitation when you do not block or use other interrupts.
The last received byte is available in the IO registers while receiving the next byte (by hardware). So, there are some milliseconds to process the byte in the fast enough ISR.