mathertel/DMXSerial #48: Simultaneous sending and receiving DMX?

chendy, 2020-08-11:
Will this library support 'simultaneous' sending and receiving? 
I want to receive DMX, remap/reprocess some values, then send DMX. Assuming the processing is very lightweight, will this library support this?

Thanks for your amazing work.

mathertel, 2020-08-13:
I am nos sure what you want so let's clarify the use-cases:

* If you want to implement DMX (not RDM) there is no device defined that implements both roles (Controller and Slave).
The DMX Serial library is made for devices that have one of these roles.

* Devices that have input and output only have a wired connection between these adapters and pass the signal through to enable daisy chaining the slave devices,

* Devices that support RDM (extension of DMX) should use the DMXSerial2 library. Here sending in slave devices is required to answer specific requests.

* Repeaters, splitters, . . .  not supported by the library. Here a "amplifier" is an option that doesn't read and writes the protocol but uses multiple connected 485 couplers to enable longer/multiple DMX lines. Very robust, no processor required.

chendy, 2020-09-05:
Thanks @mathertel 

In addition to the serial port used for the DMX by this library, would there be a spare serial port that I could use? So I could have 2x Arduinos, one set up as a DMX controller and one as a DMX slave. And use UART for communication between these two devices.

mathertel, 2020-09-06:
That seems complex and may work but I never did it.

willwatson27, 2021-01-27:
> Thanks @mathertel
> 
> In addition to the serial port used for the DMX by this library, would there be a spare serial port that I could use? So I could have 2x Arduinos, one set up as a DMX controller and one as a DMX slave. And say use UART for communication between these two devices if using a Mega. Or use say SPI to connect two ATmega328s?

Have you had any luck with this? I'm trying to forward DMX from a grandma2 with some extra interbyte delay to fix jitter issues on some Chinese fixtures.

chendy, 2021-01-27:
Hello @willwatson27 

I didn't try this approach but instead used this with success
https://github.com/ssilverman/TeensyDMX

willwatson27, 2021-01-27:
Thanks for the lead @chendy. That definitely looks like what I'm after.

mathertel, 2021-03-08:
I cannot recommend using an Arduino UNO or ATMega as there are restrictions on the available Serial ports and also the processor may not handle all the load running multiple lines because of the slow speed. The Teensy or ESP32 is a better and faster processor that may fit.

closed 2021-03-08 by mathertel

ChetNocatee, 2023-01-18:
I am currently using the DMXSerial library, version 1.5.3, with an Arduino Uno R3 to use DMXSerial.init(DMXReceiver) and DMXSerial.read  in the setup() function where I try to write an incoming DMX packet , the first 45 channels, to a byte array IChannel[k].  I then execute DMXSerial.init(DMXController) and DMXSerial.write as the last step in setup().  In the loop() function I try to augment 4 of the 45 channels in IChannel[k] through A0 and A1 with a potentiometric joystick.  The joystick adjustment passes through to TX (D1), but not the originalDMXSerial.write input from the setup() function.  I am reading both inputs and outputs with  PicoScope 2206B MSO oscilloscope. If I load IChannel[k], as shown in the attached file, the loaded array elements show up in the output, along with any joystick adjustments.  Why can I not load an array with DMXSerial.write and subsequently get it out from TX with DMXSerial.read.  I can make a schematic of the circuitry used if anyone is interested
[Manual_Spotlight_Control_2.txt](https://github.com/mathertel/DMXSerial/files/10448770/Manual_Spotlight_Control_2.txt)