mathertel/DMXSerial #49: DMX Tester which can send and recive DMX values

AnnoyedSandwich, 2020-08-25:
Hi, 

I want to build a little DMX tester box. And I have some questions.

1. I want to have a little LED turn on as soon as any DMX value is received. It should check all 512 addresses and as soon as it receives one value above 0 turns the LED on. Is this possible or do you have to specify the address it "listens" to?

2. I also want to make use of the Send function to send a test signal. How would I fade every address to 255 and back? And can I simultaneously us the Rx bin for receiving and the Tx pin for sending at the same time?

mathertel, 2021-02-26:
1 The DMXSerial always receives all bytes on the wire into the buffer. You need to loop through all buffer offsets and see when there is a value on any channel. 
2 see DMXSerialFlow.ino
3 teensy no, leonardo yes.

closed 2021-02-26 by mathertel

mathertel, 2021-06-08:
Have a look into the DMXSniff example working with a Leonardo board.
Please take the latest version from the GitHub repository.