mathertel/DMXSerial #51: Compile Error on Arduino MEGA2560

MikeWDale, 2021-03-08:
I very much want to use your DMX library.
Attached Arduino Sketch works on UNOWifi or UNO but will not compile for MEGA2560.
Basic Blink sketch works for MEGA2560; so I assume board is OK.
I simply do not understand and hope you may help.
Sketch and Error message attached.

[BoardwalkFourModesMasterDMXBlynk210209.docx](https://github.com/mathertel/DMXSerial/files/6101408/BoardwalkFourModesMasterDMXBlynk210209.docx)
[ERROR MESSAGE WHEN TRYING TO COMPILE SKETCH.docx](https://github.com/mathertel/DMXSerial/files/6101410/ERROR.MESSAGE.WHEN.TRYING.TO.COMPILE.SKETCH.docx)
+

mathertel, 2021-03-08:
This is the conflict with the Hardware - Serial Interface. 
See:

Details on how to use the library with ARDUINO MEGA and ARDUINO Leonardo boards and different ports:
http://mathertel.blogspot.de/2013/05/update-for-dmxserial-library.html

closed 2021-03-08 by mathertel

MikeWDale, 2021-03-08:
Thank you for directing me to your blog page.  Unfortunately I do not know how to uncomment the line 
#define DMX_USE_PORT1
I assume this is a line in DMXSerial.h
The Arduino IDE will not open .h files, only .ino and .pde files
Please tell me how to do this.