mathertel/DMXSerial #59: ports 2 and 3 on mega

jaygrovr, 2022-01-09:
How can I use serial ports 2  or 3 on mega 2560?  I only see documentation and notes for switching to port1.

mathertel, 2022-01-10:
The ports 2 and 3 for mega2560 has not been implemented - but it should be easy to do.
The port specific defines are in the file DMXSerial_avr.h and more ports can be added here by adding definitions.
Maybe you create a pull request for this ?

jaygrovr, 2022-01-10:
Port 0 and Port 1 work currently with your software?

jaygrovr, 2022-01-11:
So I am having enabling the "DMX_USE_PORT1" feature.  I would like to use the default or port 0 for serial monitoring the arduino mega 2560, and use port1 uart for the DMX receiver.  When I try to setup "Serial.begin(9600) on my mega I get a bunch of errors compiling, which seem like the same port is being used for dmx rx.  I have uncommented out the section in the DMXSerial_avr.h and to use_port1, and have added the line in my little program  #define DMX_USE_PORT1, but I still get the following errors 

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_25'
C:\Users\Jason\AppData\Local\Temp\arduino_build_806647\libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_26'
C:\Users\Jason\AppData\Local\Temp\arduino_build_806647\libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
Using library DMXSerial at version 1.5.2 in folder: C:\src\libraries\DMXSerial 
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

mathertel, 2022-01-13:
I just checked by the following steps:

(read documentation in https://github.com/mathertel/DmxSerial)

* Open the example **`DmxSerialRecv.ino`** with Arduino IDE.
* Change processor to “Arduino Mega or Mega 2560”
* add  `Serial.begin(57600);` in setup()
* add `Serial.println("DMXBOX_TestLED");`in setup()
.
* Open the file  **`DMXSerial_avr.h`** in the libraries / DMXSerial / src folder.
* uncomment line # 33 to activate the definition `#define DMX_USE_PORT1`
.
* Verify the Sketch – it must compile without the errors above.

Please verify that this works for you as well.

jaygrovr, 2022-01-13:
Ok I did exactly that.. followed your instructions and I got the same error.

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':(.text+0x0): multiple definition of `__vector_25'C:\Users\Jason\AppData\Local\Temp\arduino_build_658739\libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined hereHardwareSerial0.cpp.o (symbol from plugin): In function `Serial':(.text+0x0): multiple definition of `__vector_26'C:\Users\Jason\AppData\Local\Temp\arduino_build_658739\libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined herecollect2.exe: error: ld returned 1 exit statusUsing library DMXSerial at version 1.5.2 in folder: C:\homeassistant\esphome projects\homeenerymonitor-heatpump\EmonESP-master\EmonESP-master\src\libraries\DMXSerial exit status 1Error compiling for board Arduino Mega or Mega 2560.

<code>// - - - - -// DmxSerial - A hardware supported interface to DMX.// DmxSerialRecv.ino: Sample DMX application for retrieving 3 DMX values:// address 1 (red) -> PWM Port 9// address 2 (green) -> PWM Port 6// address 3 (blue) -> PWM Port 5// // Copyright (c) 2011-2015 by Matthias Hertel, http://www.mathertel.de// This work is licensed under a BSD style license. See http://www.mathertel.de/License.aspx// // Documentation and samples are available at http://www.mathertel.de/Arduino// 25.07.2011 creation of the DmxSerial library.// 10.09.2011 fully control the serial hardware register//            without using the Arduino Serial (HardwareSerial) class to avoid ISR implementation conflicts.// 01.12.2011 include file and extension changed to work with the Arduino 1.0 environment// 28.12.2011 changed to channels 1..3 (RGB) for compatibility with the DmxSerialSend sample.// 10.05.2012 added some lines to loop to show how to fall back to a default color when no data was received since some time.// - - - - -
#include <DMXSerial.h>
// Constants for demo program
const int RedPin =    9;  // PWM output pin for Red Light.const int GreenPin =  6;  // PWM output pin for Green Light.const int BluePin =   5;  // PWM output pin for Blue Light.
#define RedDefaultLevel   100#define GreenDefaultLevel 200#define BlueDefaultLevel  255
void setup () {  DMXSerial.init(DMXReceiver);  Serial.begin(57600);  Serial.println("dmx text rx");    // set some default values  DMXSerial.write(1, 80);  DMXSerial.write(2, 0);  DMXSerial.write(3, 0);    // enable pwm outputs  pinMode(RedPin,   OUTPUT); // sets the digital pin as output  pinMode(GreenPin, OUTPUT);  pinMode(BluePin,  OUTPUT);}

void loop() {  // Calculate how long no data backet was received  unsigned long lastPacket = DMXSerial.noDataSince();    if (lastPacket < 5000) {    // read recent DMX values and set pwm levels     analogWrite(RedPin,   DMXSerial.read(1));    analogWrite(GreenPin, DMXSerial.read(2));    analogWrite(BluePin,  DMXSerial.read(3));
  } else {    // Show pure red color, when no data was received since 5 seconds or more.    analogWrite(RedPin,   RedDefaultLevel);    analogWrite(GreenPin, GreenDefaultLevel);    analogWrite(BluePin,  BlueDefaultLevel);  } // if}
// End.<code>
and I edited the avr file as so
// For using the serial port 1 on a Arduino MEGA 2560 board, enable the following DMX_USE_PORT1 definition.#define DMX_USE_PORT1

mathertel, 2022-01-13:
The #define USART0_RX_vect	_VECTOR(25) is still used by the DMXSerial library so the uncomment of #define PORT1 in the file C:\homeassistant\esphome projects\homeenerymonitor-heatpump\EmonESP-master\EmonESP-master\src\libraries\DMXSerial\src\DMXSerial_avr.h is not in effect. Please check this file.
The #define USART1_RX_vect	_VECTOR(36) must be used to assign USARTn_RX_vect 
#define USARTn_RX_vect USART1_RX_vect

jaygrovr, 2022-01-13:
im confused, the define port1 is COMMENTED out.. so its not in effect.  I can delete it.  But I dont understand what you want me to do.

jaygrovr, 2022-01-13:
I did that, and I get the same error as before.
jason

 
 

Line #33 must look like this:

mathertel, 2022-01-13:
Maybe we should share screens for some minutes.
Send me an invite to ma email address directly.
You can find on https://www.mathertel.de/impressum.aspx

mathertel, 2022-01-15:
found the right file to uncomment the PORT1 directive....

closed 2022-01-15 by mathertel