mathertel/DMXSerial #60: DMX wave form is different then the actual DMX512 controller. Why??

brainstormtec, 2022-01-17:
[image]
i was sending the DMX data to the Fixture using Arduino UNO with DMXserial library but the fxture was not responding. then i tried with the DMX512 actual Controller the fixture start responding.  i have taken the waveform on ocelloscope as atteched. The Green Waveform is from the Arduino UNO and the Yellow is from Actual DMX512 Fixture. I am to be able to identify the issue. 
I have tried the same by upload the code from Arduino Cloud to my UNO the result was the same. Then i tried with the Differnt Computer with the fresh Arduino IDE installation, Still the result is same. I cant able to figure it out how to resolve ??? Please Help.

mathertel, 2022-01-18:
yes, this looks strange.
Local or cloud based IDE/programming should not make a difference.
It should be a serial signal with 56kBaud most of the time.
Please check GND / tx pin signal on the arduino first. The check that is is equal after the RS485 conversion.

brainstormtec, 2022-01-23:
> 

Hello Sir,
Here itis the graph on oscilloscope, Where Greeen is Tx and GND and Yellow is the B Output of RS485.
[image]

please guide to make it working for me.

brainstormtec, 2022-02-04:
I have tested it further,  I have got an observation that if we make an Arduino Board and Max485 as a DMX controller and another as receiver it works fine without issue, However the Waveform is not slandered DMX signal. 
I observed the Library a file DMXSerial_avr.h is there and it is written #define CalcPreScale(B) (((((F_CPU) / 8) / (B)) - 1) / 2) for scaling. i tried to find details and try to understand it but i need more explanation and elaboration regarding it. 
From where it will get "F_CPU".   
Will please someone help ???

mathertel, 2022-02-04:
This is the CPU frequence. This is calculating the "scaler" to get the baud rate right.
See 22.2 ff in the processor manual.
It seems this doesn't wok for you right. Maybe a bug.

Can you try the original code from the manual with the right baud rates and your CPU speed ?:
```
#define FOSC 1843200// Clock Speed
#define BAUD 9600
#define (MYUBRR FOSC/16/BAUD-1)
void main( void )
{...
USART_Init ( MYUBRR );
...} // main
void USART_Init( unsigned int ubrr){
/* Set baud rate */
UBRRH = (unsigned char)(ubrr>>8);
UBRRL = (unsigned char)ubrr;
/* Enable receiver and transmitter */
UCSRB = (1<<RXEN)|(1<<TXEN);
/* Set frame format: 8data, 2stop bit */
UCSRC = (1<<USBS)|(3<<UCSZ0);
} // USART_Ini
```

brainstormtec, 2022-02-05:
> 

Thank you for the reply and guidance. I will check and revert.

brainstormtec, 2022-02-12:
[image]
Sir, Try something with F_CPU different values but not succeed.
 I have some observation, the Green Graph is the DMX Tx signal using this Library, and the Yellow is the Signal from the actual DMX Controller.
Now the Graph Marked Red Part is the same but the Blue Marked Timing is different. Can you please suggest, How I can be able to increase this timing to make the graph same as DMX Controller Signal.

mathertel, 2022-02-16:
I just checked and **it works as expected** using an original Arduino MEGA 2560 board and PORT1 activated by modifying the line # 33 in DMYSerial_avr.h. Also checked the prescaler calc that is 3 at 250000 baud with 16 MHz CPU.

I don't use these boards but why not - let's see.

This is the wiring:

[image]

the solder bridges at 0 and 1 are open. Bridge at 2 to control the direction is closed. I used a slightly modified DMXSerialFlow example and also had a retail DMX device on the line.

This is the generated DMX signal. A little bit noisy but it gets decoded correctly. Trigger at break condition:

[image]

mathertel, 2022-02-16:
I close until further findings get reported. I am really interested in the root cause at your setup.

closed 2022-02-16 by mathertel