mathertel/DMXSerial #64: DMX Sniff example not compiling

stijnb1234, 2022-04-13:
I get the following error:

```
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_18'
libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_19'
libraries\DMXSerial\DMXSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```

I'm compiling for an Arduino Uno.

stijnb1234, 2022-04-13:
I'm stupid, it's mentioned in the comments above....

closed 2022-04-13 by stijnb1234

mhlsf, 2022-10-18:
I'm having the same issue, may I know the solution ? @stijnb1234

stijnb1234, 2022-10-18:
> I'm having the same issue, may I know the solution ? @stijnb1234

"This only works on boards with multiple UARTs (like the Mega2560), or with a virtual USB serial port in addition to a normal UART (like the Leonardo or Micro)."

The Arduino Uno only has one UART available.