mathertel/DMXSerial #77: Noise on one DMX channel

MicroMarty, 2023-06-02:
I have an issue using the library:

One channel contains noise mixed with dmx data, even if this data are 0. 

The problematic channel is the 31, when sending data on the first 3 channels using the DmxSerialSend example.

And the channel 511, when sending data on all 512 channels.

I tested this library on an Arduino nano and an Arduino pro mini and this issue appears with both boards.

Thanks for your help!

tommyco10, 2023-07-21:
I have this same problem too, my dmx analyser 'flicker finder' finds an intermittent flicker value of 128 on channel 511 when transmitting 512 channels.

MicroMarty, 2023-09-23:
Did you find something? 

I moved away to esp8266, I didn't find any correct DMX library on Atmega...