mathertel/DMXSerial #78: New DMX modules does not operate well anymore

ctenov, 2023-06-22:
Hi, please I have used this DMX module below years ago with DMX Simple library and a converter. But nowadays this DMX module does not operates anymore and the factory does not knows about this change of behaviour.
https://www.aliexpress.com/item/33048856643.html?spm=a2g0o.order_list.order_list_main.78.5b631802RUDtLv

I used this converter (years ago), with Arduino Mega and this was ok
https://www.aliexpress.com/item/1005005394780041.html?spm=a2g0o.productlist.main.63.6f9170eelhQcss&algo_pvid=d1a92f7d-2ca6-45a8-80d5-bfa462622112&aem_p4p_detail=202306220628262849249773006480008068791&algo_exp_id=d1a92f7d-2ca6-45a8-80d5-bfa462622112-31&pdp_npi=3%40dis%21USD%210.42%210.42%21%21%21%21%21%40211bea0816874405062642564d07b9%2112000032885489567%21sea%21BR%21160034427&curPageLogUid=USzLohK77Ld0&search_p4p_id=202306220628262849249773006480008068791_32

And now I am trying to change to this one with your serial library
https://www.aliexpress.com/item/1005001666201174.html?spm=a2g0o.productlist.main.3.6f9170eelhQcss&algo_pvid=d1a92f7d-2ca6-45a8-80d5-bfa462622112&aem_p4p_detail=202306220628262849249773006480008068791&algo_exp_id=d1a92f7d-2ca6-45a8-80d5-bfa462622112-1&pdp_npi=3%40dis%21USD%211.95%211.95%21%21%21%21%21%40211bea0816874405062642564d07b9%2112000017042933857%21sea%21BR%21160034427&curPageLogUid=Ks8Vwk74o23p&search_p4p_id=202306220628262849249773006480008068791_2

You could say me that this new module will work ???

Many thanks

mathertel, 2023-06-26:
Do you have more information on the components and schema of this adapter ?

ctenov, 2023-06-27:
Maybe here you could see the components:
https://www.aliexpress.com/item/1005005115538257.html?spm=a2g0o.productlist.main.7.6d8256b7Rx6P6b&algo_pvid=9487aa76-29c6-4c65-9dba-87048e6d90cb&aem_p4p_detail=202306270703582409604221026440001013419&algo_exp_id=9487aa76-29c6-4c65-9dba-87048e6d90cb-3&pdp_npi=3%40dis%21USD%215.0%214.9%21%21%21%21%21%40212272e216878746381224838d07cb%2112000031710474437%21sea%21BR%21160034427&curPageLogUid=tJ0Lex8Xz484&search_p4p_id=202306270703582409604221026440001013419_4

And here the Schema:
https://www.aliexpress.com/item/1005004116648223.html?spm=a2g0o.order_list.order_list_main.47.5c92194dKiCGKW

Have you any information about S1-DR DMX modules that appears to be faulty nowadays ?