mathertel/DMXSerial #80: ESP8266 compatible?

kleinmantara, 2023-08-25:
Hi,

in the issue https://github.com/mathertel/DMXSerial/issues/75 there is the [comment](https://github.com/mathertel/DMXSerial/issues/75#issuecomment-1537465344) "I use it on Arduino UNO, ESP8266, ESP32."

I have the same problem @blockiyt, i got an compile error in platform io for an espressif8266:
```
In file included from lib/DMXSerial/src/DMXSerial.cpp:13:
lib/DMXSerial/src/DMXSerial.h:39:10: fatal error: avr/io.h: No such file or directory
   39 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
```

blockiyt, 2023-09-07:
Seems that ESP8266 is not compatible with avr.
I've written you an email, with an example how you can do that.

mathertel, 2023-09-10:
The issue #75 was updated.

The DMXSerial library only supports the CPUs mentioned in the [README.md](https://github.com/mathertel/DMXSerial/blob/master/README.md) file.
See above correction.

Sorry for any confusions,

closed 2023-09-10 by mathertel

kleinmantara, 2023-09-11:
This library is available for the ESP8266 here: https://github.com/claudeheintz/LXESP8266DMX/